abperiasamy/rtl8812AU_8821AU_linux #257: May I reawaken #241 ?   Major issues with timers,  Kernel 4.15, breaking (for example) Wifi drivers...

mikeh007, 2018-07-11:
I see that  the issue here  https://github.com/abperiasamy/rtl8812AU_8821AU_linux/issues/241

- is marked closed although it's clearly tripping up lots of non-experts like myself.      Here was my comment / question from that thread:   

As a very keen supporter of Linux, and a user of Ubuntu and Lubuntu, but not an expert developer, can I just ask -- innocently -- how is it that, a mainline kernel is released with such radical changes as to timer code, which (it would seem) then SCREWS UP untold numbers of legacy drivers for things like WiFi adaptors ?

I've loved Ubuntu for 7 years now and run my entire business on it - desktops and laptops - yesterday I upgraded my main laptop from 16.04.02 to 18.04 to then find the Wifi as flaky as hell.

Spent all last evening, and half this morning, learning about drivers - mine is for an innocent, excellent, $10 USB wireless adaptor from TPLINK), and on about the 20th attempt to 'fix the driver module' for the RTL8192 which this device uses - now using TPLINK's own linux driver-install instructions --- I find the sudo make install step FALLS OVER, because.. well from one point of view, pardon me, but Kernel 4.15 seems not fit for purpose ? ie it WILL NOT ALLOW me to use a driver created by the manufacturer and which plainly has worked fine on 16.04.

I have the highest regard for the kernel developer,   but didn't this sort of thing ought to be checked out BEFORE urging us all to upgrade ? 

16.04 was the last LTS, and 18.04 this is the next: I assumed it would be a safe move up. Evidently not.    Resolved to have to backtrack to a reinstall of 16.04LTS where I was before and lose 2 days for no progress.   Is that where this kernel leaves us all ?

I do hope the kernel is fit for purpose again by the time 16.04 LTS ends.

Thanks all - any and all advice and wisdom would be much appreciated.

Mike H
London

mikeh007, 2018-07-11:
Update on the above query:   I have now found a solution,  thanks to  Mange at  
https://github.com/Mange/rtl8192eu-linux-driver.   

I'll close this and leave it for the record,   I found the above after about finding a lot of other historically-usable driver fixes,  but all (except this one above)  proved to be incompatible with kernel 4.15, due to these timer changes.

closed 2018-07-11 by mikeh007

ItsTehStory, 2018-08-04:
Hey (I'm going to try it anyway since I'm desperate) but, correct me if I'm wrong, are you saying the rtl8192 driver works for the rtl8812/rtl8821? That would be great actually.

ljschumacher, 2018-08-06:
I suspect not, as mikeh007 said their device uses the rtl8192 - did you get it to work for the 8821?

ljschumacher, 2018-08-06:
@ItsTehStory using the dkms install instructions, making sure no other drivers were installed, I now got my rtl8821au working on Ubuntu 18.04

ItsTehStory, 2018-08-06:
Yeah @ljschumacher I got it working thanks!