abperiasamy/rtl8812AU_8821AU_linux #262: CentOS 7 compile errors, kernel 3.10.0-862

a31k, 2018-09-03:
```make[1]: Entering directory `/usr/src/kernels/3.10.0-862.el7.x86_64'
  CC [M]  /tmp/t4u/rtl8812AU_8821AU_linux-master/core/rtw_cmd.o
  CC [M]  /tmp/t4u/rtl8812AU_8821AU_linux-master/core/rtw_security.o
  CC [M]  /tmp/t4u/rtl8812AU_8821AU_linux-master/core/rtw_debug.o
  CC [M]  /tmp/t4u/rtl8812AU_8821AU_linux-master/core/rtw_io.o
  CC [M]  /tmp/t4u/rtl8812AU_8821AU_linux-master/core/rtw_ioctl_query.o
  CC [M]  /tmp/t4u/rtl8812AU_8821AU_linux-master/core/rtw_ioctl_set.o
  CC [M]  /tmp/t4u/rtl8812AU_8821AU_linux-master/core/rtw_ieee80211.o
  CC [M]  /tmp/t4u/rtl8812AU_8821AU_linux-master/core/rtw_mlme.o
  CC [M]  /tmp/t4u/rtl8812AU_8821AU_linux-master/core/rtw_mlme_ext.o
  CC [M]  /tmp/t4u/rtl8812AU_8821AU_linux-master/core/rtw_wlan_util.o
  CC [M]  /tmp/t4u/rtl8812AU_8821AU_linux-master/core/rtw_vht.o
  CC [M]  /tmp/t4u/rtl8812AU_8821AU_linux-master/core/rtw_pwrctrl.o
  CC [M]  /tmp/t4u/rtl8812AU_8821AU_linux-master/core/rtw_rf.o
  CC [M]  /tmp/t4u/rtl8812AU_8821AU_linux-master/core/rtw_recv.o
  CC [M]  /tmp/t4u/rtl8812AU_8821AU_linux-master/core/rtw_sta_mgt.o
  CC [M]  /tmp/t4u/rtl8812AU_8821AU_linux-master/core/rtw_ap.o
  CC [M]  /tmp/t4u/rtl8812AU_8821AU_linux-master/core/rtw_xmit.o
  CC [M]  /tmp/t4u/rtl8812AU_8821AU_linux-master/core/rtw_p2p.o
  CC [M]  /tmp/t4u/rtl8812AU_8821AU_linux-master/core/rtw_tdls.o
  CC [M]  /tmp/t4u/rtl8812AU_8821AU_linux-master/core/rtw_br_ext.o
  CC [M]  /tmp/t4u/rtl8812AU_8821AU_linux-master/core/rtw_iol.o
  CC [M]  /tmp/t4u/rtl8812AU_8821AU_linux-master/core/rtw_sreset.o
  CC [M]  /tmp/t4u/rtl8812AU_8821AU_linux-master/core/rtw_btcoex.o
  CC [M]  /tmp/t4u/rtl8812AU_8821AU_linux-master/core/rtw_beamforming.o
  CC [M]  /tmp/t4u/rtl8812AU_8821AU_linux-master/core/rtw_odm.o
  CC [M]  /tmp/t4u/rtl8812AU_8821AU_linux-master/core/efuse/rtw_efuse.o
  CC [M]  /tmp/t4u/rtl8812AU_8821AU_linux-master/os_dep/osdep_service.o
  CC [M]  /tmp/t4u/rtl8812AU_8821AU_linux-master/os_dep/linux/os_intfs.o
/tmp/t4u/rtl8812AU_8821AU_linux-master/os_dep/linux/os_intfs.c:809:2: error: initialization from incompatible pointer type [-Werror]
  .ndo_select_queue = rtw_select_queue,
  ^
/tmp/t4u/rtl8812AU_8821AU_linux-master/os_dep/linux/os_intfs.c:809:2: error: (near initialization for ‘rtw_netdev_ops.<anonymous>.ndo_select_queue’) [-Werror]
cc1: all warnings being treated as errors
make[2]: *** [/tmp/t4u/rtl8812AU_8821AU_linux-master/os_dep/linux/os_intfs.o] Error 1
make[1]: *** [_module_/tmp/t4u/rtl8812AU_8821AU_linux-master] Error 2
make[1]: Leaving directory `/usr/src/kernels/3.10.0-862.el7.x86_64'
make: *** [modules] Error 2
```
Error when trying to `make` on CentOS 7 after `make clean`.

DavidLP, 2018-09-13:
Same issue on xubuntu 14.04 LTS with 3.13 kernel

a31k, 2018-09-13:
@DavidLP I think there's an issue with older kernels as they (presumably) use an older compiler. I have installed Ubuntu 18.04 and am currently using the driver successfully after taking the steps I have outlined in the following gist: https://gist.github.com/a31k/4ece1b8160823544395176f8f23fdd0c
Hope this helps.