abperiasamy/rtl8812AU_8821AU_linux #265: Error in kernel 4.19

darkdragon-001, 2018-10-23:
In latest stable kernel 4.19, the following `#define` changed from

    #define IEEE80211_MAX_AMPDU_BUF 0x40

to

    #define IEEE80211_MAX_AMPDU_BUF_HT      0x40
    #define IEEE80211_MAX_AMPDU_BUF         0x100

Just changing it doesn't seem enough. Can anyone help me to get it working on 4.19? It works without any problems on 4.18...

darkdragon-001, 2018-10-23:
I found a [fix](https://github.com/gordboy/rtl8812au/commit/f92c90364f21a7d9ea2d38ea85bb9d6287111a6f). Please apply it!

SleepyMario, 2018-10-31:
Same here, problems on 4.19, working fine on 4.18

iblech, 2018-10-31:
I can confirm that the fix linked by @darkdragon-001 works fine. It didn't appy cleanly, though; a version which does it [located here](https://www.speicherleck.de/iblech/stuff/0001-Update-to-kernel-4.19.patch).

foggy1, 2018-11-06:
+1 to darkdragon/iblech's patch

nunomgsantos80, 2018-11-16:
How can i fix it to resolve the problem?

nunomgsantos80, 2018-11-18:
https://github.com/greearb/rtl8812AU_8821AU_linux it works

scinart, 2018-12-15:
Confirmed, https://github.com/greearb/rtl8812AU_8821AU_linux it works

chaotaklon, 2019-03-07:
added support for linux v4.20
https://github.com/chaotaklon/rtl8812AU_8821AU_linux