abperiasamy/rtl8812AU_8821AU_linux #266: HalPwrSeqCmdParsing: FAil to polling Offset[0x5]=01

psj900918, 2018-10-26:
I'm trying to use NIC TPLink AC1300 T4U in Ubuntu 16.04.
(kernel version : 4.4.0-138-generic)

At first time, i followed the instruction and it does not recognize my NIC (after reboot, after network-manager restart).
I found out adding USB_DEVICE id manually (which is 2357:0115 for mine) in ./os_dep/linux/usb_intf.c can let the driver register mine, so i did make / make install / dkms install things again
(i did 'make clean', 'dkms remove' things before reinstall).
Now, it seems partially recognized,  which results like

(dmesg)
RTL871X: HalPwrSeqCmdParsing: Fail to polling Offset[0x5]=01
RTL871X: HalPwrSeqCmdParsing: Fail to polling Offset[0x5]=01
RTL871X: HalPwrSeqCmdParsing: Fail to polling Offset[0x5]=01
RTL871X: HalPwrSeqCmdParsing: Fail to polling Offset[0x5]=01
...

 - lsusb does not shows the device name of my NIC.
 - in 'ifconfig', no wireless device is listed
 - in 'iwconfig', a wireless device appears like

wlx00e04c881200 unassociated Nickname:"<WIFI@REALTEK>"
           Mode:Managed  Frequency=2.412GHz  Access Point: Not-Associated
...

Wondering why this driver does not work fine with my NIC.

rrdelaney, 2019-01-06:
Having the same issue :(

danpospisil, 2019-03-17:
The same here :/

danpospisil, 2019-03-18:
Ok, the problem is that V3 https://wikidevi.com/wiki/TP-LINK_Archer_T4U_v3 doesnt have RTL8812AU but RTL8812BU so this driver will not work and I believe the {USB_DEVICE(0x2357, 0x0115),.driver_info = RTL8812}, /* TP-Link - Archer T4U AC1300 */ shouldn't be listed in usb_intf.c. 
I found a working driver here: https://github.com/cilynx/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959

drgnomage, 2019-04-10:
@danpospisil Thank you! I've been trying the wrong drivers for days and your comment lead me to looking at the box and realising the seller had sent me a v3 and not the v1 I bought!