abperiasamy/rtl8812AU_8821AU_linux #271: openwrt

greearb, 2018-11-08:


greearb, 2018-11-08:
This patch lets one more easily compile the driver against latest openwrt.
[0002-makefile-Add-entry-for-compiling-against-openwrt-tre.patch.txt](https://github.com/abperiasamy/rtl8812AU_8821AU_linux/files/2563359/0002-makefile-Add-entry-for-compiling-against-openwrt-tre.patch.txt)

nunomgsantos80, 2018-11-16:
Hi, any news? Thks

greearb, 2018-11-16:
I got it working fine with some more patches...please see my repo.

nunomgsantos80, 2018-11-18:
@greearb thks, it works. Nice job

md3kung, 2018-11-29:
@greearb I can't figure out how to build it on my MIPS device ... What did you use in your OpenWrt Makefile to make it work ?

greearb, 2018-11-30:
I noticed today that I cannot get this to associate with WPA or WPA2.  Open seems to work.

Are you able to associate to an encrypted network?

Thanks,
Ben

greearb, 2018-12-04:
This was a bug in my supplicant...I have fixed it now and WPA2 works fine.

Thanks,
Ben

gojpdchx, 2018-12-16:
It looks like this driver has been added to OpenWrt, but I believe that users will not be able to use RTL8812AU / RTL8821AU / RTL8811AU for 802.11ac AP without patching the driver. zebulon2/rtl8812au@ae6a22a + zebulon2/rtl8812au@566ea44 should do the work.

Note that Realtek released several versions of rtl8812au driver, and the above commits are based on the version 5.1.5. It is the last known version to support all three chipsets, RTL8812AU / RTL8821AU / RTL8811AU.
Version 5.2.20 is also available, but it only supports RTL8812AU (zebulon2/rtl8812au-driver-5.2.20).

kimocoder, 2018-12-16:
You are wrong. We've got most versions of the drivers, 5.1.5, v5.2.20(.2) and v5.3.4 with lots of patches added and support for all 3 chipsets on all drivers after adding HAL from the v5.1.5 to them.

https://github.com/aircrack-ng/rtl8812au

gojpdchx, 2018-12-17:
@kimocoder 
Many thanks for the information. It seems that the rtl8812au driver by aircrack-ng is the best maintained. AP mode on RTL8811AU with 5.3.4 works fine, and I get 200 - 240 Mbps / 150 - 190 Mbps (Download / Upload speed).

May I know if you (or other developers) are interested in submitting the driver upstream?
It looks like the driver is similar to [rtl8188eu in staging](https://github.com/torvalds/linux/tree/master/drivers/staging/rtl8188eu).

greearb, 2018-12-17:
On 12/16/18 10:12 PM, gojpdchx wrote:
> @kimocoder <https://github.com/kimocoder>
> Many thanks for the information. It seems that the rtl8812au driver by aircrack-ng is the best maintained. AP mode on RTL8811AU with 5.3.4 works fine, and I [...]

I found it very difficult to determine which was the 'best' 8812au github project to work on.

Maybe the various projects could sync up so that there was one main driver repository,
and ones that are old and un-maintained could at least make a note that they are deprecated?

At least the abperiasamy tree is unlikely to be accepted into Linux in my opinion, except perhaps in
the staging tree.

Thanks,
Ben

-- 
Ben Greear <greearb@candelatech.com>
Candela Technologies Inc  http://www.candelatech.com

gojpdchx, 2018-12-18:
@greearb 
Yes, there are many abandoned 8812au projects, and some do not have the 802.11ac AP patch (including the abperiasamy tree, so your fork and lede-project/source@d8f861f are probably affected).
Even Ubuntu's official rtl8812au package uses [rsalveti/rtl8812au](https://github.com/rsalveti/rtl8812au
), which seems to be abandoned.

@kimocoder actively maintains the [aircrack-ng/rtl8812au](https://github.com/aircrack-ng/rtl8812au) tree, so I think he can consider submitting it upstream.