abperiasamy/rtl8812AU_8821AU_linux #273: Does not compile on Sparklinux (Debian Stretch)

famewolf, 2018-11-17:
Fails with following:

make[1]: Entering directory '/usr/src/linux-headers-4.19.2-sparky-686-pae'
  CC [M]  /home/famewolf/rtl8812AU_8821AU_linux/core/rtw_cmd.o
In file included from /home/famewolf/rtl8812AU_8821AU_linux/include/drv_types.h:35,
                 from /home/famewolf/rtl8812AU_8821AU_linux/core/rtw_cmd.c:22:
/home/famewolf/rtl8812AU_8821AU_linux/include/wifi.h:980: error: "IEEE80211_MAX_AMPDU_BUF" redefined [-Werror]
 #define IEEE80211_MAX_AMPDU_BUF 0x40
 
In file included from ./include/net/cfg80211.h:24,
                 from /home/famewolf/rtl8812AU_8821AU_linux/include/osdep_service_linux.h:88,
                 from /home/famewolf/rtl8812AU_8821AU_linux/include/osdep_service.h:41,
                 from /home/famewolf/rtl8812AU_8821AU_linux/include/drv_types.h:32,
                 from /home/famewolf/rtl8812AU_8821AU_linux/core/rtw_cmd.c:22:
./include/linux/ieee80211.h:1442: note: this is the location of the previous definition
 #define IEEE80211_MAX_AMPDU_BUF  0x100
 
cc1: all warnings being treated as errors
make[2]: *** [scripts/Makefile.build:306: /home/famewolf/rtl8812AU_8821AU_linux/core/rtw_cmd.o] Error 1
make[1]: *** [Makefile:1517: _module_/home/famewolf/rtl8812AU_8821AU_linux] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.19.2-sparky-686-pae'
make: *** [Makefile:1584: modules] Error 2

nunomgsantos80, 2018-11-21:
Hi,
Try this fork:
https://github.com/greearb/rtl8812AU_8821AU_linux

shadow1163, 2019-01-29:
@nunomgsantos80 , it works, thank you very much.

antivanov, 2019-03-15:
@nunomgsantos80 Thanks, worked for me on Manjaro 4.19.28-1-MANJARO