abperiasamy/rtl8812AU_8821AU_linux #278: Attempting to compile on 4.15.0-43-generic #46-Ubuntu

Skrappjaw, 2019-01-10:
I bought this 11 dollar dongle for my moms PC.... Cannot compile.
make returns with...

> make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.15.0-43-generic/build M=/home/davi/rtl8812AU_8821AU_linux-master  modules
> make[1]: Entering directory '/usr/src/linux-headers-4.15.0-43-generic'
>   CC [M]  /home/davi/rtl8812AU_8821AU_linux-master/core/rtw_cmd.o
>   CC [M]  /home/davi/rtl8812AU_8821AU_linux-master/core/rtw_security.o
>   CC [M]  /home/davi/rtl8812AU_8821AU_linux-master/core/rtw_debug.o
>   CC [M]  /home/davi/rtl8812AU_8821AU_linux-master/core/rtw_io.o
>   CC [M]  /home/davi/rtl8812AU_8821AU_linux-master/core/rtw_ioctl_query.o
>   CC [M]  /home/davi/rtl8812AU_8821AU_linux-master/core/rtw_ioctl_set.o
>   CC [M]  /home/davi/rtl8812AU_8821AU_linux-master/core/rtw_ieee80211.o
>   CC [M]  /home/davi/rtl8812AU_8821AU_linux-master/core/rtw_mlme.o
>   CC [M]  /home/davi/rtl8812AU_8821AU_linux-master/core/rtw_mlme_ext.o
>   CC [M]  /home/davi/rtl8812AU_8821AU_linux-master/core/rtw_wlan_util.o
>   CC [M]  /home/davi/rtl8812AU_8821AU_linux-master/core/rtw_vht.o
>   CC [M]  /home/davi/rtl8812AU_8821AU_linux-master/core/rtw_pwrctrl.o
>   CC [M]  /home/davi/rtl8812AU_8821AU_linux-master/core/rtw_rf.o
>   CC [M]  /home/davi/rtl8812AU_8821AU_linux-master/core/rtw_recv.o
>   CC [M]  /home/davi/rtl8812AU_8821AU_linux-master/core/rtw_sta_mgt.o
>   CC [M]  /home/davi/rtl8812AU_8821AU_linux-master/core/rtw_ap.o
>   CC [M]  /home/davi/rtl8812AU_8821AU_linux-master/core/rtw_xmit.o
>   CC [M]  /home/davi/rtl8812AU_8821AU_linux-master/core/rtw_p2p.o
>   CC [M]  /home/davi/rtl8812AU_8821AU_linux-master/core/rtw_tdls.o
>   CC [M]  /home/davi/rtl8812AU_8821AU_linux-master/core/rtw_br_ext.o
>   CC [M]  /home/davi/rtl8812AU_8821AU_linux-master/core/rtw_iol.o
>   CC [M]  /home/davi/rtl8812AU_8821AU_linux-master/core/rtw_sreset.o
>   CC [M]  /home/davi/rtl8812AU_8821AU_linux-master/core/rtw_btcoex.o
>   CC [M]  /home/davi/rtl8812AU_8821AU_linux-master/core/rtw_beamforming.o
>   CC [M]  /home/davi/rtl8812AU_8821AU_linux-master/core/rtw_odm.o
>   CC [M]  /home/davi/rtl8812AU_8821AU_linux-master/core/efuse/rtw_efuse.o
>   CC [M]  /home/davi/rtl8812AU_8821AU_linux-master/os_dep/osdep_service.o
>   CC [M]  /home/davi/rtl8812AU_8821AU_linux-master/os_dep/linux/os_intfs.o
>   CC [M]  /home/davi/rtl8812AU_8821AU_linux-master/os_dep/linux/usb_intf.o
>   CC [M]  /home/davi/rtl8812AU_8821AU_linux-master/os_dep/linux/usb_ops_linux.o
>   CC [M]  /home/davi/rtl8812AU_8821AU_linux-master/os_dep/linux/ioctl_linux.o
>   CC [M]  /home/davi/rtl8812AU_8821AU_linux-master/os_dep/linux/xmit_linux.o
>   CC [M]  /home/davi/rtl8812AU_8821AU_linux-master/os_dep/linux/mlme_linux.o
>   CC [M]  /home/davi/rtl8812AU_8821AU_linux-master/os_dep/linux/recv_linux.o
>   CC [M]  /home/davi/rtl8812AU_8821AU_linux-master/os_dep/linux/ioctl_cfg80211.o
> /home/davi/rtl8812AU_8821AU_linux-master/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_cfg80211_indicate_sta_assoc’:
> /home/davi/rtl8812AU_8821AU_linux-master/os_dep/linux/ioctl_cfg80211.c:3334:3: error: ISO C90 forbids mixed declarations and code [-Werror=declaration-after-statement]
>    u8 ie_offset;
>    ^~
> cc1: all warnings being treated as errors
> scripts/Makefile.build:332: recipe for target '/home/davi/rtl8812AU_8821AU_linux-master/os_dep/linux/ioctl_cfg80211.o' failed
> make[2]: *** [/home/davi/rtl8812AU_8821AU_linux-master/os_dep/linux/ioctl_cfg80211.o] Error 1
> Makefile:1551: recipe for target '_module_/home/davi/rtl8812AU_8821AU_linux-master' failed
> make[1]: *** [_module_/home/davi/rtl8812AU_8821AU_linux-master] Error 2
> make[1]: Leaving directory '/usr/src/linux-headers-4.15.0-43-generic'
> Makefile:1584: recipe for target 'modules' failed
> make: *** [modules] Error 2

gojpdchx, 2019-01-11:
How about aircrack-ng's repo, https://github.com/aircrack-ng/rtl8812au/tree/v5.3.4 ?
This repository seems to be unmaintained for some time.

michalpiekarski, 2019-01-11:
https://github.com/abperiasamy/rtl8812AU_8821AU_linux/commit/076ac48d9457e947f50835b38dfc616ac9ab210b#r31908013

Skrappjaw, 2019-01-12:
> How about aircrack-ng's repo, https://github.com/aircrack-ng/rtl8812au/tree/v5.3.4 ?
> This repository seems to be unmaintained for some time.
> 
> Also, what locale are you using?
> Example:
> 
> ```
> [user@pc user]$ locale
> LANG=en_US.UTF-8
> LC_CTYPE=en_US.UTF-8
> LC_NUMERIC="en_US.UTF-8"
> LC_TIME=en_US.UTF-8
> ...
> ```

LANG=en_US.UTF-8
LANGUAGE=
LC_CTYPE="en_US.UTF-8"
LC_NUMERIC="en_US.UTF-8"
LC_TIME="en_US.UTF-8"
LC_COLLATE="en_US.UTF-8"
LC_MONETARY="en_US.UTF-8"
LC_MESSAGES="en_US.UTF-8"
LC_PAPER="en_US.UTF-8"
LC_NAME="en_US.UTF-8"
LC_ADDRESS="en_US.UTF-8"
LC_TELEPHONE="en_US.UTF-8"
LC_MEASUREMENT="en_US.UTF-8"
LC_IDENTIFICATION="en_US.UTF-8"
LC_ALL=

chaotaklon, 2019-03-07:
added support for linux v4.20
https://github.com/chaotaklon/rtl8812AU_8821AU_linux

closed 2019-03-11 by Skrappjaw