abperiasamy/rtl8812AU_8821AU_linux #298: USB device ID {USB_DEVICE(0x2357, 0x0115) should be removed from this driver

Baudouin0, 2019-07-03:
This device is RTL8812BU and is not compatible with the 8812AU.
Should work with rtl8822bu driver

.{USB_DEVICE(0x2357, 0x0115), .driver_info = RTL8822B}, /* Archer USB T4Uv3 */

Lynaex, 2019-07-09:
Thanks for your answer Baudouin
But neither of them.. I made a mistake, I was searching rtl8812au (RTL8811-AC600-MWA255)
Recently I got the solution here:
https://forums.opensuse.org/showthread.php/536637-RTL8811-AC600-MWA255?p=2908084#post2908084
Regards

faultylee, 2019-08-20:
I recently purchased a TP-Link T4U v3 with `USB_DEVICE(0x2357, 0x0115)` and I've tested all the 8812au/8821au drivers, non worked accept 8822b, confirming what @Baudouin0 is stating