abperiasamy/rtl8812AU_8821AU_linux #301: Build error on Arch Linux. Kernel 5.1.

danwerspb, 2019-07-17:
qwer@sev-lucky  /tmp/rtl8812AU_8821AU_linux   master  sudo make -f Makefile.dkms install                                 
make clean
make[1]: вход в каталог «/tmp/rtl8812AU_8821AU_linux»
make ARCH=arm64 CROSS_COMPILE= -C /usr/src/linux-headers-5.1.18-1-MANJARO-ubuntu18.04_aarch64/kernel-4.9 M=/tmp/rtl8812AU_8821AU_linux clean
make[2]: вход в каталог «/tmp/rtl8812AU_8821AU_linux»
make[2]: *** /usr/src/linux-headers-5.1.18-1-MANJARO-ubuntu18.04_aarch64/kernel-4.9: Нет такого файла или каталога.  Останов.
make[2]: выход из каталога «/tmp/rtl8812AU_8821AU_linux»
make[1]: *** [Makefile:1643: clean] Ошибка 2
make[1]: выход из каталога «/tmp/rtl8812AU_8821AU_linux»
make: *** [Makefile.dkms:11: src_install] Ошибка 2

Previously, the assembly took place normally

reavertm, 2019-07-21:
Try:

sed -e "s/^\(CONFIG_PLATFORM_.* = \).*/\1n/g" -i Makefile || die
sed -e "s/^\(CONFIG_PLATFORM_I386_PC = \).*/\1y/g" -i Makefile || die

danwerspb, 2019-07-25:
It works. Thanks

closed 2019-07-25 by danwerspb