abperiasamy/rtl8812AU_8821AU_linux #302: Unabel to install driver for Ubuntu 18.04

anilsathyan7, 2019-07-19:
Does it support ubuntu 18.04??
Iam getting an error in make command

```
/tmp/t4u/rtl8812AU_8821AU_linux-master$ make
make ARCH=arm64 CROSS_COMPILE= -C /usr/src/linux-headers-4.15.0-54-generic-ubuntu18.04_aarch64/kernel-4.9 M=/tmp/t4u/rtl8812AU_8821AU_linux-master  modules
make[1]: *** /usr/src/linux-headers-4.15.0-54-generic-ubuntu18.04_aarch64/kernel-4.9: No such file or directory.  Stop.
Makefile:1622: recipe for target 'modules' failed
make: *** [modules] Error 2

```

reavertm, 2019-07-21:
Unfortunately defaults has been changed in Makefile for reasons unknown.

sed -e "s/^\(CONFIG_PLATFORM_.* = \).*/\1n/g" -i Makefile || die
sed -e "s/^\(CONFIG_PLATFORM_I386_PC = \).*/\1y/g" -i Makefile || die

should do the trick

harshavardhana, 2019-09-22:
Feel free to send a fix @reavertm