abperiasamy/rtl8812AU_8821AU_linux #304: Build error on Ubuntu Disco Dingo Kernel 5.0.0-23-generic

slvrshot, 2019-08-01:
I run the following commands: 

```
git clone https://github.com/abperiasamy/rtl8812AU_8821AU_linux.git
cd rtl8812au-master
make 

ake ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.0.0-23-generic/build M=/home/jay/Downloads/rtl8812au-master  modules
make[1]: Entering directory '/usr/src/linux-headers-5.0.0-23-generic'
  CC [M]  /home/jay/Downloads/rtl8812au-master/os_dep/linux/os_intfs.o
/home/jay/Downloads/rtl8812au-master/os_dep/linux/os_intfs.c:1069:22: error: initialization of ‘u16 (*)(struct net_device *, struct sk_buff *, struct net_device *, u16 (*)(struct net_device *, struct sk_buff *, struct net_device *))’ {aka ‘short unsigned int (*)(struct net_device *, struct sk_buff *, struct net_device *, short unsigned int (*)(struct net_device *, struct sk_buff *, struct net_device *))’} from incompatible pointer type ‘u16 (*)(struct net_device *, struct sk_buff *, void *, u16 (*)(struct net_device *, struct sk_buff *, struct net_device *))’ {aka ‘short unsigned int (*)(struct net_device *, struct sk_buff *, void *, short unsigned int (*)(struct net_device *, struct sk_buff *, struct net_device *))’} [-Werror=incompatible-pointer-types]
  .ndo_select_queue = rtw_select_queue,
                      ^~~~~~~~~~~~~~~~
/home/jay/Downloads/rtl8812au-master/os_dep/linux/os_intfs.c:1069:22: note: (near initialization for ‘rtw_netdev_ops.ndo_select_queue’)
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:286: /home/jay/Downloads/rtl8812au-master/os_dep/linux/os_intfs.o] Error 1
make[1]: *** [Makefile:1606: _module_/home/jay/Downloads/rtl8812au-master] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.0.0-23-generic'
make: *** [Makefile:1064: modules] Error 2

```
I'd appreciate any help. Thanks!

jairelee, 2019-08-17:
This seems to be working in Linux Kernel 5.0.0-25-generic

harshavardhana, 2019-09-22:
Closing as it's working on 25-generic

closed 2019-09-22 by harshavardhana