abperiasamy/rtl8812AU_8821AU_linux #312: WiFi Authentication broken after fa68771

toolboc, 2019-10-10:
Changeset fa68771 seems to have broken WPA authentication on Nvidia Jetson Nano devices.  I have installed the rtl8812AU driver on 30+ Nano devices and most recently pulled in latest changes on a new devices.  I am able to successfully build and install with:

```
sudo apt install dkms
git clone https://github.com/abperiasamy/rtl8812AU_8821AU_linux
cd rtl8812AU_8821AU_linux
sed -i 's/CONFIG_PLATFORM_I386_PC = y/CONFIG_PLATFORM_I386_PC = n/g' Makefile
sed -i 's/CONFIG_PLATFORM_ARM_JET_NANO = n/CONFIG_PLATFORM_ARM_JET_NANO = y/g' Makefile
sudo make -f Makefile.dkms install
```

However, any attempts to authenticate to my home network via WPA2 fail after installation containing this latest change.  Reverting changeset fa68771 and rebuilding/reinstalling allows for successful authentication.  The changeset in question specifically affects WIFI_AUTH and is definitely the culprit.  This should probably be investigated further as the issue likely affects other devices/chipsets.

kondratev, 2019-11-11:
try this pull https://github.com/abperiasamy/rtl8812AU_8821AU_linux/pull/322

guilamo, 2019-11-20:
Hello, I had the same problem. Thanks for you fix kondratev but it is not enough : before I was not able to connect at all and now I can connect but only automatically after a reboot.

 On start up the wifi now automatically connects to my internet network. However if I stop the connection I face the same problem of authentication I had before and I have to reboot to reconnect.

guilamo, 2019-11-20:
Edit : (edit button doesn't work for some reason)
I work on an Ubuntu 18.04 desktop PC not on a Jetson nano.
I tried kondratev pull and reverting fa68771 changeset and had the same results (connect only on system start).

greearb, 2019-11-21:
I saw failures on the 5.2 kernel, and reverting fa68771 appears to have fixed the problem.  I'm using the NIC in station mode, so I am not sure if the pull #322 would actually apply?  I'm using WUSB6300 NIC.

kondratev, 2019-11-22:
> I saw failures on the 5.2 kernel, and reverting [fa68771](https://github.com/abperiasamy/rtl8812AU_8821AU_linux/commit/fa68771376a637c0b5f9cfa53da008570939a259) appears to have fixed the problem. [...]

I using my "COMFAST (Model: CF-912AC) as AP too.

poliander, 2019-12-21:
I can confirm that the aforementioned commit breaks AP mode on my Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac 2T2R DB WLAN Adapter.

netskink, 2020-02-17:
Im trying to use this driver with Debian 10 stable.  It fails for me. It says authentication password is incorrect. I did a detailed write up here.  It has the same error connecting as a client as it does when I try to use it as a wifi access point.  You can see the wifi access point, but password fails.

https://gist.github.com/netskink/5f92db32c4c2587211664673d236ef0b

netskink, 2020-02-17:
I pulled the git commit before the fa68771, built the driver and it works.  I could connect to wifi clients and setup an access point using the usb wifi dongle.

romanrm, 2020-03-22:
Unfortunately this is still a problem 5 months after being reported. The breaking commit is still in master, and the rollback pull request #324 reverting it still doesn't seem to be merged. Just spent a few hours more on trying to get this driver to work than was actually necessary; and every user is likely to go through the same process during all this time.

Only using client mode, not AP. Client auth to BS did not work at all until fa68771 was reverted.

Benzer1406, 2020-05-08:
I got it working doing the following steps:
**Download the Repo**
1. `mkdir -p /tmp/t4u`
2. `cd /tmp/t4u`
3. `git clone https://github.com/abperiasamy/rtl8812AU_8821AU_linux.git`

**Checkout to working commit**
4. `git checkout 4235b0ec7d7220a6364586d8e25b1e8cb99c36f1` //corresponds to commit 4235b0e

**Installation**
5. `cd rtl8812AU_8821AU_linux-master`
6. `sudo apt install linux-headers-$(uname -r)`
7. `make`
8. `sudo make install`

**Loading the module**
9. `sudo modprobe -a rtl8812au`

Please be aware that you probably have to restart your computer.