abperiasamy/rtl8812AU_8821AU_linux #317: New kernel,

devomaa, 2019-10-26:
When I try to install this, it fails, and i have version 5.3.7 so it must be kernel version

devomaa, 2019-10-26:
sorry about editing so many times, i am not good at markdown

jskovjyskebankdk, 2019-10-28:
See https://github.com/abperiasamy/rtl8812AU_8821AU_linux/pull/316

devomaa, 2019-10-28:
> See #316

I tried compiling and installing the driver from your fork in Arch linux kernel 5.3.7 and unfortunalely I got same crash

jskovjyskebankdk, 2019-10-29:
Maybe you did not use the kernel-5.3 branch?
That is my only input, I am afraid.

devomaa, 2019-10-29:
Yes, I compiled the correct branch 5.3... at Latest commit 822b485 - 4 days ago 

first uninstalled the driver, then cloned your repo, then compiled and installed the driver. Reboot PC just in case anything wrong in memory or whatever. Finally attached the Wifi dongle, and at dmesg got again that crash log and non functional dongle.

jskovjyskebankdk, 2019-10-30:
Sorry, no idea then.

karlkleinpaste, 2019-11-05:
From #316, the small fallthrough patch doesn't apply, there is no "case WIFI_ASSOCREQ" in the file at all.
I applied the 2nd patch just fine and used usual dkms procedure to re-install. The driver no longer causes failure w/stack trace but it won't associate with any AP, either.

On modprobe, it goes through typical recognition & registration, then NetworkManager gets told about it, which goes through typical stages resulting in "managed/disconnected" state. Then, trying to make a connection results in repeating sequences of rejection:

Nov  5 07:12:03 godiva wpa_supplicant[3012]: wlan1: Trying to associate with 08:bd:43:59:5f:84 (SSID='FEMA Nuclear Mediation' freq=2412 MHz)
Nov  5 07:12:04 godiva kernel: RTL871X: rtw_set_802_11_connect(wlan1)  fw_state=0x00000008
Nov  5 07:12:04 godiva NetworkManager[1483]: <info>  [1572955924.3218] device (wlan1): supplicant interface state: scanning -> associating
Nov  5 07:12:04 godiva NetworkManager[1483]: <info>  [1572955924.3219] device (p2p-dev-wlan1): supplicant management interface state: scanning -> associating
Nov  5 07:12:04 godiva kernel: RTL871X: start auth
Nov  5 07:12:05 godiva wpa_supplicant[3012]: wlan1: CTRL-EVENT-ASSOC-REJECT status_code=1
Nov  5 07:12:05 godiva wpa_supplicant[3012]: wlan1: CTRL-EVENT-ASSOC-REJECT status_code=1
Nov  5 07:12:05 godiva NetworkManager[1483]: <info>  [1572955925.8961] device (wlan1): supplicant interface state: associating -> disconnected

Note CTRL-EVENT-ASSOC-REJECT code 1. This happens several times, during which NM pops a dialog to ask for password (it is correct, has not changed, and other wifi devices connect to this AP using that password just fine), eventually failing on taking too long:

Nov  5 07:12:24 godiva kernel: RTL871X: sta recv deauth reason code(0) sta:08:bd:43:59:5f:84, ignore = 0
Nov  5 07:12:25 godiva NetworkManager[1483]: <warn>  [1572955945.8215] device (wlan1): Activation: (wifi) association took too long

No concrete suggestions to offer of my own, I do networks but it's been a very long time since I poked seriously at any wifi driver.