abperiasamy/rtl8812AU_8821AU_linux #326: Doesn't work on Ubuntu 19.10 kernel 5.3.0-24-generic

sean-huni, 2020-01-06:
The WiFi device (TP-Link AC1200 Wireless Dual-Band USB Adapter) doesn't work on the latest Ubuntu 19.10 on the latest kernel.

`➜  rtl8812AU_8821AU_linux git:(master) ✗ cd ~/aTpLink/rtl8812AU_8821AU_linux      
➜  rtl8812AU_8821AU_linux git:(master) ✗ 
➜  rtl8812AU_8821AU_linux git:(master) ✗ sudo make -f Makefile.dkms install
make clean
make[1]: Entering directory '/home/sean/aTpLink/rtl8812AU_8821AU_linux'
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.3.0-24-generic/build M=/home/sean/aTpLink/rtl8812AU_8821AU_linux clean
make[2]: Entering directory '/usr/src/linux-headers-5.3.0-24-generic'
  CLEAN   /home/sean/aTpLink/rtl8812AU_8821AU_linux/Module.symvers
make[2]: Leaving directory '/usr/src/linux-headers-5.3.0-24-generic'
make[1]: Leaving directory '/home/sean/aTpLink/rtl8812AU_8821AU_linux'
mkdir -p '/usr/src/rtl8812au-4.3.14'
cp -r dkms.conf Kconfig Makefile.dkms Makefile platform core hal include os_dep '/usr/src/rtl8812au-4.3.14'
cp Makefile '/usr/src/rtl8812au-4.3.14/Makefile'
sed 's/#MODULE_VERSION#/4.3.14/' dkms.conf > '/usr/src/rtl8812au-4.3.14/dkms.conf'
dkms add -m rtl8812au -v 4.3.14 2>/dev/null || true
dkms build -m rtl8812au -v 4.3.14
Module rtl8812au/4.3.14 already built for kernel 5.3.0-24-generic/4
dkms install -m rtl8812au -v 4.3.14
Module rtl8812au/4.3.14 already installed on kernel 5.3.0-24-generic/x86_64
➜  rtl8812AU_8821AU_linux git:(master) ✗ sudo make install
install -p -m 644 rtl8812au.ko  /lib/modules/5.3.0-24-generic/kernel/drivers/net/wireless/
install: cannot stat 'rtl8812au.ko': No such file or directory
make: *** [Makefile:1628: install] Error 1
➜  rtl8812AU_8821AU_linux git:(master) ✗ sudo make clean
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.3.0-24-generic/build M=/home/sean/aTpLink/rtl8812AU_8821AU_linux clean
make[1]: Entering directory '/usr/src/linux-headers-5.3.0-24-generic'
make[1]: Leaving directory '/usr/src/linux-headers-5.3.0-24-generic'
➜  rtl8812AU_8821AU_linux git:(master) ✗ sudo make      
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.3.0-24-generic/build M=/home/sean/aTpLink/rtl8812AU_8821AU_linux  modules
make[1]: Entering directory '/usr/src/linux-headers-5.3.0-24-generic'
  CC [M]  /home/sean/aTpLink/rtl8812AU_8821AU_linux/core/rtw_cmd.o
  CC [M]  /home/sean/aTpLink/rtl8812AU_8821AU_linux/core/rtw_security.o
  CC [M]  /home/sean/aTpLink/rtl8812AU_8821AU_linux/core/rtw_debug.o
  CC [M]  /home/sean/aTpLink/rtl8812AU_8821AU_linux/core/rtw_io.o
  CC [M]  /home/sean/aTpLink/rtl8812AU_8821AU_linux/core/rtw_ioctl_query.o
  CC [M]  /home/sean/aTpLink/rtl8812AU_8821AU_linux/core/rtw_ioctl_set.o
  CC [M]  /home/sean/aTpLink/rtl8812AU_8821AU_linux/core/rtw_ieee80211.o
  CC [M]  /home/sean/aTpLink/rtl8812AU_8821AU_linux/core/rtw_mlme.o
  CC [M]  /home/sean/aTpLink/rtl8812AU_8821AU_linux/core/rtw_mlme_ext.o
  CC [M]  /home/sean/aTpLink/rtl8812AU_8821AU_linux/core/rtw_wlan_util.o
  CC [M]  /home/sean/aTpLink/rtl8812AU_8821AU_linux/core/rtw_vht.o
  CC [M]  /home/sean/aTpLink/rtl8812AU_8821AU_linux/core/rtw_pwrctrl.o
  CC [M]  /home/sean/aTpLink/rtl8812AU_8821AU_linux/core/rtw_rf.o
  CC [M]  /home/sean/aTpLink/rtl8812AU_8821AU_linux/core/rtw_recv.o
  CC [M]  /home/sean/aTpLink/rtl8812AU_8821AU_linux/core/rtw_sta_mgt.o
  CC [M]  /home/sean/aTpLink/rtl8812AU_8821AU_linux/core/rtw_ap.o
  CC [M]  /home/sean/aTpLink/rtl8812AU_8821AU_linux/core/rtw_xmit.o
  CC [M]  /home/sean/aTpLink/rtl8812AU_8821AU_linux/core/rtw_p2p.o
  CC [M]  /home/sean/aTpLink/rtl8812AU_8821AU_linux/core/rtw_tdls.o
  CC [M]  /home/sean/aTpLink/rtl8812AU_8821AU_linux/core/rtw_br_ext.o
  CC [M]  /home/sean/aTpLink/rtl8812AU_8821AU_linux/core/rtw_iol.o
  CC [M]  /home/sean/aTpLink/rtl8812AU_8821AU_linux/core/rtw_sreset.o
  CC [M]  /home/sean/aTpLink/rtl8812AU_8821AU_linux/core/rtw_btcoex.o
  CC [M]  /home/sean/aTpLink/rtl8812AU_8821AU_linux/core/rtw_beamforming.o
  CC [M]  /home/sean/aTpLink/rtl8812AU_8821AU_linux/core/rtw_odm.o
  CC [M]  /home/sean/aTpLink/rtl8812AU_8821AU_linux/core/efuse/rtw_efuse.o
  CC [M]  /home/sean/aTpLink/rtl8812AU_8821AU_linux/os_dep/osdep_service.o
  CC [M]  /home/sean/aTpLink/rtl8812AU_8821AU_linux/os_dep/linux/os_intfs.o
  CC [M]  /home/sean/aTpLink/rtl8812AU_8821AU_linux/os_dep/linux/usb_intf.o
  CC [M]  /home/sean/aTpLink/rtl8812AU_8821AU_linux/os_dep/linux/usb_ops_linux.o
  CC [M]  /home/sean/aTpLink/rtl8812AU_8821AU_linux/os_dep/linux/ioctl_linux.o
  CC [M]  /home/sean/aTpLink/rtl8812AU_8821AU_linux/os_dep/linux/xmit_linux.o
  CC [M]  /home/sean/aTpLink/rtl8812AU_8821AU_linux/os_dep/linux/mlme_linux.o
  CC [M]  /home/sean/aTpLink/rtl8812AU_8821AU_linux/os_dep/linux/recv_linux.o
  CC [M]  /home/sean/aTpLink/rtl8812AU_8821AU_linux/os_dep/linux/ioctl_cfg80211.o
  CC [M]  /home/sean/aTpLink/rtl8812AU_8821AU_linux/os_dep/linux/rtw_cfgvendor.o
  CC [M]  /home/sean/aTpLink/rtl8812AU_8821AU_linux/os_dep/linux/wifi_regd.o
  CC [M]  /home/sean/aTpLink/rtl8812AU_8821AU_linux/os_dep/linux/rtw_android.o
[... 61 lines cut ...]
  CC [M]  /home/sean/aTpLink/rtl8812AU_8821AU_linux/platform/platform_ops.o
  CC [M]  /home/sean/aTpLink/rtl8812AU_8821AU_linux/core/rtw_mp.o
  CC [M]  /home/sean/aTpLink/rtl8812AU_8821AU_linux/core/rtw_mp_ioctl.o
  CC [M]  /home/sean/aTpLink/rtl8812AU_8821AU_linux/core/rtw_bt_mp.o
  LD [M]  /home/sean/aTpLink/rtl8812AU_8821AU_linux/rtl8812au.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /home/sean/aTpLink/rtl8812AU_8821AU_linux/rtl8812au.mod.o
  LD [M]  /home/sean/aTpLink/rtl8812AU_8821AU_linux/rtl8812au.ko
make[1]: Leaving directory '/usr/src/linux-headers-5.3.0-24-generic'
➜  rtl8812AU_8821AU_linux git:(master) ✗ sudo make install
install -p -m 644 rtl8812au.ko  /lib/modules/5.3.0-24-generic/kernel/drivers/net/wireless/
/sbin/depmod -a 5.3.0-24-generic
➜  rtl8812AU_8821AU_linux git:(master) ✗ sudo modprobe -a rtl8812au
➜  rtl8812AU_8821AU_linux git:(master) ✗  sudo apt install linux-headers-$(uname -r)
Reading package lists... Done
Building dependency tree       
Reading state information... Done
linux-headers-5.3.0-24-generic is already the newest version (5.3.0-24.26).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
➜  rtl8812AU_8821AU_linux git:(master) ✗ 
➜  rtl8812AU_8821AU_linux git:(master) ✗ 
➜  rtl8812AU_8821AU_linux git:(master) ✗ 
➜  rtl8812AU_8821AU_linux git:(master) ✗ 
`

sean-huni, 2020-01-06:
It works well on the Ubuntu 18.

weilun23, 2020-01-27:
i'm on 5.3.0-26-generic and the below link worked for me:

https://github.com/aircrack-ng/rtl8812au/tree/v5.3.4

lexterror, 2020-02-14:
weilun23: The link you provided does not work well for me. It let me connect to the internet, but I experienced low bandwidth and constant disconnects. Hopefully with time it will be resolved.

sean-huni, 2020-03-28:
> i'm on 5.3.0-26-generic and the below link worked for me:
> 
> https://github.com/aircrack-ng/rtl8812au/tree/v5.3.4

Thank you @weilun23 it works perfectly fine for me, with the latest linux ubuntu kernel.

closed 2020-03-28 by sean-huni