abperiasamy/rtl8812AU_8821AU_linux #331: Wifi authentication does not work

guidomeijer, 2020-03-11:
I pulled from master and installed as instructed but when I try to connect to a network it tells me the password is not correct (while it is). I'm on Ubuntu 18.04 using a TP-LINK Archer T4U AC1300 adapter. The device is recognized but dmesg tells me 
`IPv6: ADDRCONF(NETDEV_UP): wlx18d6c71aa922: link is not ready`
Any help would be appreciated!

romanrm, 2020-03-22:
Check out the issue #312

Benzer1406, 2020-05-08:
I got it working doing the following steps:
**Download the Repo**
1. `mkdir -p /tmp/t4u`
2. `cd /tmp/t4u`
3. `git clone https://github.com/abperiasamy/rtl8812AU_8821AU_linux.git`

**Checkout to working commit**
4. `git checkout 4235b0ec7d7220a6364586d8e25b1e8cb99c36f1` //corresponds to commit 4235b0e

**Installation**
5. `cd rtl8812AU_8821AU_linux-master`
6. `sudo apt install linux-headers-$(uname -r)`
7. `make`
8. `sudo make install`

**Loading the module**
9. `sudo modprobe -a rtl8812au`

Please be aware that you probably have to restart your computer.