abperiasamy/rtl8812AU_8821AU_linux #336: TPLink Archer-T4U Not working after driver installation Ubuntu 20.04

JoshuaBragg, 2020-08-05:
I was able to successfully build and install the drivers following the installation steps listed below both for Ubuntu and also tried with DKMS. The wifi adapter still fails to be detected and I have tried all I could think of :/

The device appears in the output of `sudo lsusb` as `ID 2357:0115 TP-Link 802.11ac NIC`

It does not appear on `sudo lshw -C network`

`uname -r` returns `5.4.0-42-generic`

The building with `make` and installing with `depmod` returned successfully and running `lsmod` displays `rtl8812au            1347584  0` in the list of drivers.

I am not sure what else to try :( I previously had Ubuntu 18.04 and this same wifi adapter worked for me after using the beta drivers from TPLink website. This is a fresh installation of Ubuntu on a new disk so there are no leftover remnants of the 18.04 installation on this drive.

Any help on what I can do to make my adapter work would be much appreciated!

Townsheriff, 2020-11-19:
I'm having exactly same issue. In dmesg I can see a stack trace:
```
[ 4377.253494] RTL871X: module init start
[ 4377.253497] RTL871X: rtl8812au v4.3.14_13455.20150212_BTCOEX20150128-51
[ 4377.253497] RTL871X: rtl8812au BT-Coex version = BTCOEX20150128-51
[ 4378.324822] RTL871X: invalid offset:0x42 
[ 4379.566885] ------------[ cut here ]------------
[ 4379.566913] WARNING: CPU: 0 PID: 90790 at net/wireless/core.c:867 wiphy_register+0x6d1/0x930 [cfg80211]
[ 4379.566914] Modules linked in: rtl8812au(OE+) cfg80211 btrfs xor zstd_compress raid6_pq ufs qnx4 hfsplus hfs minix ntfs msdos jfs xfs libcrc32c rndis_host cdc_ether usbnet mii cdc_acm hid_logitech_hidpp joydev hid_logitech_dj edac_mce_amd nls_iso8859_1 snd_hda_codec_hdmi uas input_leds usb_storage snd_hda_codec_realtek snd_hda_codec_generic ledtrig_audio snd_hda_intel snd_intel_dspcfg snd_hda_codec snd_hda_core snd_hwdep snd_pcm snd_seq_midi nouveau snd_seq_midi_event kvm snd_rawmidi crct10dif_pclmul ghash_clmulni_intel mxm_wmi snd_seq video ttm aesni_intel drm_kms_helper crypto_simd i2c_algo_bit fb_sys_fops snd_seq_device cryptd glue_helper snd_timer syscopyarea sysfillrect sysimgblt snd wmi_bmof ccp k10temp soundcore mac_hid sch_fq_codel parport_pc ppdev lp parport drm ip_tables x_tables autofs4 hid_generic usbhid hid crc32_pclmul r8169 i2c_piix4 ahci nvme realtek libahci nvme_core wmi gpio_amdpt gpio_generic [last unloaded: rtl8812au]
[ 4379.566970] CPU: 0 PID: 90790 Comm: modprobe Tainted: G        W  OE     5.4.0-51-generic #56-Ubuntu
[ 4379.566971] Hardware name: To Be Filled By O.E.M. To Be Filled By O.E.M./B450M Pro4, BIOS P1.60 11/05/2018
[ 4379.566990] RIP: 0010:wiphy_register+0x6d1/0x930 [cfg80211]
[ 4379.566994] Code: ff e9 78 fa ff ff 0f 0b b8 ea ff ff ff e9 6c fa ff ff 0f 0b b8 ea ff ff ff e9 60 fa ff ff 0f 0b b8 ea ff ff ff e9 54 fa ff ff <0f> 0b b8 ea ff ff ff e9 48 fa ff ff 0f 0b b8 ea ff ff ff e9 3c fa
[ 4379.566995] RSP: 0018:ffffa31103e9f970 EFLAGS: 00010246
[ 4379.566997] RAX: ffffffffc0fea7f0 RBX: ffff94b565b96b00 RCX: ffffffffc0fea7e0
[ 4379.566999] RDX: ffffffffc0fea880 RSI: 0000000000000001 RDI: 000000000000001e
[ 4379.567000] RBP: ffffa31103e9f9f0 R08: 0000000000000004 R09: 0000000000000003
[ 4379.567001] R10: 0000000000000cc0 R11: ffffa31200738fc5 R12: ffff94b565b96b00
[ 4379.567002] R13: ffff94b606fdd000 R14: ffff94b5f94be830 R15: ffff94b5f94be830
[ 4379.567004] FS:  00007f37083e8540(0000) GS:ffff94b60ec00000(0000) knlGS:0000000000000000
[ 4379.567005] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[ 4379.567006] CR2: 000009710813c014 CR3: 00000003c7b36000 CR4: 00000000003406f0
[ 4379.567008] Call Trace:
[ 4379.567060]  ? _rtw_malloc+0x2d/0x2f [rtl8812au]
[ 4379.567104]  ? _rtw_memcpy+0x10/0x12 [rtl8812au]
[ 4379.567151]  ? rtw_5g_rates_init+0x1a/0x1c [rtl8812au]
[ 4379.567195]  ? rtw_spt_band_alloc+0xb0/0xb2 [rtl8812au]
[ 4379.567239]  rtw_wdev_alloc+0x107/0x2ad [rtl8812au]
[ 4379.567282]  rtw_usb_if1_init+0x138/0x205 [rtl8812au]
[ 4379.567324]  rtw_drv_init+0x244/0x2ce [rtl8812au]
[ 4379.567329]  usb_probe_interface+0xed/0x290
[ 4379.567333]  really_probe+0x2b3/0x3e0
[ 4379.567336]  driver_probe_device+0xbc/0x100
[ 4379.567338]  device_driver_attach+0x5d/0x70
[ 4379.567341]  __driver_attach+0x8f/0x150
[ 4379.567343]  ? device_driver_attach+0x70/0x70
[ 4379.567346]  bus_for_each_dev+0x7e/0xc0
[ 4379.567348]  driver_attach+0x1e/0x20
[ 4379.567350]  bus_add_driver+0x152/0x1f0
[ 4379.567353]  driver_register+0x74/0xd0
[ 4379.567356]  usb_register_driver+0x89/0x130
[ 4379.567358]  ? 0xffffffffc08fe000
[ 4379.567385]  rtw_drv_entry+0x86/0x1000 [rtl8812au]
[ 4379.567390]  do_one_initcall+0x4a/0x1fa
[ 4379.567393]  ? kmem_cache_alloc_trace+0x16c/0x240
[ 4379.567398]  do_init_module+0x62/0x250
[ 4379.567401]  load_module+0x10b8/0x1200
[ 4379.567406]  __do_sys_finit_module+0xbe/0x120
[ 4379.567408]  ? __do_sys_finit_module+0xbe/0x120
[ 4379.567413]  __x64_sys_finit_module+0x1a/0x20
[ 4379.567415]  do_syscall_64+0x57/0x190
[ 4379.567420]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
[ 4379.567422] RIP: 0033:0x7f370852d89d
[ 4379.567425] Code: 00 c3 66 2e 0f 1f 84 00 00 00 00 00 90 f3 0f 1e fa 48 89 f8 48 89 f7 48 89 d6 48 89 ca 4d 89 c2 4d 89 c8 4c 8b 4c 24 08 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d c3 f5 0c 00 f7 d8 64 89 01 48
[ 4379.567426] RSP: 002b:00007ffcca219998 EFLAGS: 00000246 ORIG_RAX: 0000000000000139
[ 4379.567428] RAX: ffffffffffffffda RBX: 000056513e0b2e70 RCX: 00007f370852d89d
[ 4379.567429] RDX: 0000000000000000 RSI: 000056513df46358 RDI: 0000000000000003
[ 4379.567430] RBP: 0000000000040000 R08: 0000000000000000 R09: 0000000000000000
[ 4379.567431] R10: 0000000000000003 R11: 0000000000000246 R12: 000056513df46358
[ 4379.567432] R13: 0000000000000000 R14: 000056513e0b2fa0 R15: 000056513e0b2e70
[ 4379.567436] ---[ end trace f159dad98dc609d0 ]---
[ 4379.697315] usb 1-4: reset high-speed USB device number 17 using xhci_hcd
[ 4379.866707] usbcore: registered new interface driver rtl8812au
[ 4379.866709] RTL871X: module init ret=0
```

enjoykcc456, 2020-12-13:
Hi @JoshuaBragg , would you mind to share how you managed to get it works in Ubuntu 18.04? Thank you.

JoshuaBragg, 2020-12-14:
@enjoykcc456 https://www.tp-link.com/us/support/download/archer-t4u/#Driver This is the driver page from TP-Link I used the drivers from here to make the card function on 18.04. Once I upgraded to 20.04 the manufacturer driver no longer worked. I didn't end up finding a way to make this card work for 20.04 so I purchased a different wifi card from D-Link :(

nestorac, 2020-12-16:
I have the same problem, Ubuntu 20.04 LTS, this driver, last version from git:

2001:3318 D-Link Corp. 11ac adapter (D-Link DWA-172)

Dmesg returns a backtrace like yours.