abperiasamy/rtl8812AU_8821AU_linux #343: Compilation fails on Manjaro Linux kernel 5.8.18-1-MANJARO

antivanov, 2020-11-13:
`make` fails with the following error:

```bash
rtl8812AU_8821AU_linux/include/rtw_security.h:236:8: error: redefinition of ‘struct sha256_state’
  236 | struct sha256_state {
      |        ^~~~~~~~~~~~
In file included from ./include/linux/filter.h:23,
                 from ./include/net/sock.h:59,
                 from ./include/net/inet_sock.h:22,
                 from ./include/net/ip.h:27,
                 from /home/anton/Downloads/rtl8812AU_8821AU_linux/core/rtw_br_ext.c:24:
./include/crypto/sha.h:84:8: note: originally defined here
   84 | struct sha256_state {
      |        ^~~~~~~~~~~~
```

tiix95, 2020-11-22:
Same from debian sid, kernel 5.9.0-3
ii  linux-image-5.9.0-3-amd64             5.9.9-1                         amd64        Linux 5.9 for 64-bit PCs (signed)
ii  linux-headers-5.9.0-3-common 5.9.9-1      all          Common header files for Linux 5.9.0-3

stamasd, 2021-05-23:
I have the exact same error on Funtoo Linux 1.4, kernel 5.10.28_p1 (the current debian_sources), gcc 9.2.0, glibc 2.33

bjinwright, 2021-06-15:
Did anyone figure this out? @tiix95 @antivanov

31337-4554551n, 2021-08-03:
Same issue on ubuntu with kernel 5.8.0-63-generic